superxslam/SuperOdom #13: There is a problem with the library dependencies.

dhfsu, 2025-05-15:
https://goen.win/auto/devpn232738/f41d8573
Hello author, I would like to ask why this problem occurs when compiling after installing the required libraries directly without using a Docker environment. As I checked, the TBB library on my computer is already the latest version.

adrienzhh, 2025-05-29:
Hi @dhfsu, make sure you are using GTSAM 4.0. 4.3 is the newest version. You can download GTSAM 4.0 using command below:
```
git clone https://github.com/borglab/gtsam.git && \
cd gtsam && git checkout 4abef92 &&\
mkdir build && \
cd build && \
cmake  \
          -DGTSAM_USE_SYSTEM_EIGEN=ON \
          -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF \
           .. && \
 make -j6 && \
  sudo make install
```

shibowing, 2025-06-11:
Hi all, feel free to let me know if the problem exists. ^ ^ 

When I compiled the code without using the docker environments, I suffer following issues and could be fixed by using following solution. 

## 🔧 Solutions for Common Issues

### 🛠️ GTSAM Shared Library Error

**Error Message:**
```bash
error while loading shared libraries: libgtsam.so.4: cannot open shared object file: No such file or directory
````

**Cause:**
This error occurs when the system cannot locate the `libgtsam.so.4` shared library, usually because it's not properly installed or not in the system's library search path.

**Solution:**

1. **Ensure GTSAM is Installed Properly**
   Navigate to the GTSAM build directory and install it:

   ```bash
   cd /root/gtsam/build
   sudo make install
   ```

2. **Update Library Search Path**
   Add the GTSAM library path to your `LD_LIBRARY_PATH` environment variable:

   ```bash
   export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
   ```

   > ✅ *Tip:* To make this permanent, add the above line to your `~/.bashrc` or Dockerfile.

```

shibowing, 2025-06-11:
Feel free to reopen the issues, if it still exists.

closed 2025-06-11 by shibowing