superxslam/SuperOdom #17: GTSAM version issue

dhfsu, 2025-06-20:
[image]

[image]

Hello author, I downloaded GTSAM according to the process described in the readme and switched to the 4abef92 commit, but there was still an error about the mismatch between the TBB library and the GTSAM version. When I checked the installed GTSAM version in the system, it eventually linked to 4.3a0. However, directly downloading the 4.1 version of GTSAM results in a compilation error. Could you please provide some suggestions to help me successfully compile it?

shibowing, 2025-06-22:
Hi! Thanks for testing!  Could you please tell me which TBB version and Ubuntu system you are using to test our code?

dhfsu, 2025-06-22:
[image]
 Hello,author,My Ubuntu version is 22.04, and the installed TBB library version is 2021.5.0-7ubuntu2.

shibowing, 2025-06-23:
Thank you! Did you use our docker? or you are trying to build the code directly on the computer?

adrienzhh, 2025-06-24:
It looks to me the issue is version mismatch between gtsam and tbb. You have not successfully linked to the gtsam 4.1. Make sure you run `sudo make install` at the end. 

```
git clone https://github.com/borglab/gtsam.git
cd gtsam && git checkout 4abef92 
mkdir build 
cd build 
cmake  
    -DGTSAM_USE_SYSTEM_EIGEN=ON 
      -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF 
       .. 
make -j6 
sudo make install
```