superxslam/SuperOdom #20: a higher frame rate

andrewliang01, 2025-06-25:
I was able to run superodom smoothly with mid360, very good open source work, but I found that the frame rate of the odometer is only 10HZ, is there a higher frame rate to choose from？

[image]

shibowing, 2025-06-25:
Hi @andrewliang01 , we also output imu odometry which is 200hz. Do you think it will solve your problem?  The lidar odometry is 10 hz because lidar scan from livox driver is 10 hz. Feel free to let me know if you have more questions.

andrewliang01, 2025-06-26:
> 嗨，我们还输出 200hz 的 imu 里程计。您认为它会解决您的问题吗？激光雷达里程计为 10 Hz，因为 livox 驱动程序的激光雷达扫描为 10 Hz。如果您有更多问题，请随时告诉我。

I need accurate and high frame rate odometry output. I am concerned that the 200Hz IMU odometry accuracy may not meet our requirements. Is there a high frame rate odometry output that combines radar and IMU?