superxslam/SuperOdom #21: ouster bag files

gglaspell, 2025-06-27:
While I was successful running Superodom with the demo [dataset](https://drive.google.com/drive/folders/1WOTj4j9t5LkKkdajFlj6bZcdmPcsJipz?usp=sharing)  and in gazebo with a simulated tb3; I am having issues running Superodom on my bag files. My bag files contain an Ouster lidar and imu topics. A majority of the time the process stops at:

`[imu_preintegration_node-3] IMU Initialization Process Finish! `

A few times I received the following errors:

```
[feature_extraction_node-1] [WARN] [1751049074.604590695] [feature_extraction_node]: meas_start_time: 1675690782.882720 > lidar_start_time: 1675690715.768898 
[feature_extraction_node-1] [WARN] [1751049074.604611274] [feature_extraction_node]: sync unsuccessfull, skipping scan frame
[feature_extraction_node-1] [WARN] [1751049074.902962772] [feature_extraction_node]: throw laser scan, only should happen at the beginning
[feature_extraction_node-1] [WARN] [1751049074.903046754] [feature_extraction_node]: removed the lidarBuf size  0, measureBuf size  2000 
[feature_extraction_node-1] [WARN] [1751049074.903063910] [feature_extraction_node]: meas_start_time: 1675690782.882720 > lidar_start_time: 1675690716.068755 
[feature_extraction_node-1] [WARN] [1751049074.903089293] [feature_extraction_node]: sync unsuccessfull, skipping scan frame
```

Please advise, I would like to get superodom running with my datasets. Also, the RVIZ confidence indicators are very nice!!

shibowing, 2025-07-01:
Hi @ShaoshuSu, I think you also have the same solutions. Feel free to share the solution here. Thank you!

shibowing, 2025-07-10:
Hi @gglaspell , I recommend to check if the time offset between IMU and Lidar message are constant value or not. If it is constant value, we can write a function to online estimate the time offset. Then apply following for each lidar scan message:  

corrected lidar stamp= imu timestamp + offset between lidar and imu stamp

shibowing, 2025-07-22:
Hi @gglaspell, we wrote a tutorial on how to set the Ouster driver. Please check it. ^^

[time-sync-ouster-lidars (2).zip](https://github.com/user-attachments/files/21361666/time-sync-ouster-lidars.2.zip)

gglaspell, 2025-07-28:
@shibowing I can confirm setting PTP works. I also can firm setting ouster to `TIME_FROM_ROS_TIME` works as well and may be a solution for @TannerGilbert