superxslam/SuperOdom #28: Feature extraction node failure to find time and map saving feature

lianrongbaoz, 2025-07-28:
Hi, i am encountering an issue where the feature extraction node is unable to find 'time' although i can see it when i ros2 topic echo lidar_points

[image]

Also, is there a map saving function that is available to load the same map for localisation after mapping?

shibowing, 2025-07-30:
Hi @tejasps28, could you help to check this function?

tejasps28, 2025-08-05:
Hi @lianrongbaoz can you share more info about your  data?

lianrongbaoz, 2025-08-11:
These are the issues i encountered:

The node doesnt run fully even when the lidar is publishing pointcloud2 and imu is running:

[image]

when the node manages to run, it outputs these warnings:

[imu_preintegration_node-3] [WARN] [1754904846.377892759] [imu_preintegration_node]: Update failed due to underconstrained call to isam2 in imuPreintegration
[imu_preintegration_node-3] [WARN] [1754904846.377971827] [imu_preintegration_node]: failureDetected
[feature_extraction_node-1] Failed to find match for field 'time'.
[feature_extraction_node-1] Failed to find match for field 'time'.
[feature_extraction_node-1] Failed to find match for field 'time'.
[feature_extraction_node-1] Failed to find match for field 'time'.
[feature_extraction_node-1] Failed to find match for field 'time'.
[feature_extraction_node-1] Failed to find match for field 'time'.
[feature_extraction_node-1] Failed to find match for field 'time'.
[feature_extraction_node-1] Failed to find match for field 'time'.

Also, it would be great to have a map saving function to reuse back the map for localisation after mapping. Thanks!

closed 2025-09-13 by shibowing