superxslam/SuperOdom #39: Adding Livox Avia Support

celikemir, 2025-09-17:
Hi,

Thank you for open-sourcing this great work. I see that this repo currently supports 3D spinning LiDAR only. I’m planning to add support for the Livox Avia, which is not a spinning LiDAR. Could you share any knowledge or insights on whether it would work? Have you had any experience with this subject?

shibowing, 2025-09-18:
Hi @celikemir,  Thanks for the questions. 

the repo also supports non-spinning lidar. Please check following config 
https://github.com/superxslam/SuperOdom/blob/ros2/super_odometry/config/livox_mid360.yaml

celikemir, 2025-09-18:
@shibowing , as datasheet say mid360 rotating mirror hybrid-solid. I was asking for livox avia like lidar. they have constant FOV.

shibowing, 2025-11-05:
feel free to reopen if you still have issues.

closed 2025-11-05 by shibowing